c-klinger/esp8266-arduino-wifirgb #2: Software on ESP just crashes when opening /ui or /admin

majuss, 2018-10-23:
I successfully compiled and flashed everything (Board Version 2.4.2, ArduinoJSON 5.1). But as soon as I go to `IP/ui` or `IP/admin`. The software just crashes completely. Which version of the above mentioned libs did you used? With 2.3 of the board software it can't compile the same applies with ArduinoJSON beta 6.x.

Attached is a decoded stacktrace when I try to open the admin panel.
[image]

I tried another board and chip, but it's showing the exact same issues. Could it be a bad solder joint or something?

c-klinger, 2018-10-23:
I just complied the code for an NodeMCU 0.9 with esp8266 v.2.4.2 and ArduinoJSON 5.13.3 and have no issue calling the /ui or /admin.
Flash Settings: 160 MHz, 4M (1M SPIDDS, v2 Lower Memory

Is the REST-API working for you? Have you tried to add additional Serial Outputs to check where the exception occurs? It seems the code is not reaching the server.on callbacks in line 66/69.

Booth /admin and /ui should be working without any connected items as they are just displaying html interfaces. So likely no bad solder (at least not from you...). How is your ESP8266 powered?

c-klinger, 2018-10-23:
Another possible solution: Please check this branch https://github.com/c-klinger/esp8266-arduino-wifirgb/tree/bugfix/esp-crashes-on-ui-or-admin ... i improved the code a regarding memory management.

majuss, 2018-10-23:
Works! Thanks a lot!!

c-klinger, 2018-10-24:
thanks for the feedback. i merged the changes to the main branch,

closed 2018-10-24 by c-klinger