c-klinger/esp8266-arduino-wifirgb #10: Huge issue with analogWrite resolution - only 1/4 of possible brightness

majuss, 2019-07-27:
So I experimented with the analogWrite Pins of an Arduino and an ESP8266. When I wrote 255 on the ESP I only measured 0.824 V, that got me wondering what is wrong... So it turns out that the analogWrite resolution on the ESP is 10-bit wide! This means the highest value is 1023 not 255 (https://circuits4you.com/2017/12/21/esp8266-pwm-example/)! So in conclusion with your sketch we are losing 3/4 of the possible brightness. This needs to get fixed asap, if you don't have any time, let me know it I will open a PR.

c-klinger, 2019-07-28:
I need to setup some hardware to reproduce/test this. I'm familiar with descriped behaivor that analogWrite resolution is 0 to 1023 by default using an ESP.

Due that fact in line 57 of the sketch 
`analogWriteRange(255);`
should adjust the range to rgb range. See https://arduino-esp8266.readthedocs.io/en/2.5.2/reference.html?highlight=analogWriteRange#analog-output

It seemed easier to use this than always converting the inputs. Have you already tested with values higher than 255 and got more light?

majuss, 2019-07-29:
Ahhh I'am so sorry I missed this line in your code and thought you only use 255 through your sketch. I measured the voltage at the gate and it is completely correct at 3.28 V. I should not open issues if I tinker in my 32°C apartment... Sorry to bother you!

closed 2019-07-29 by majuss