c-klinger/esp8266-arduino-wifirgb #11: ArduinoJson Migrating from version 5 to 6

KiWyFox, 2020-01-08:
Hello 

can you please Help me to update your code to arduinoJson V6.x.x ?

I tried the instructions from https://arduinojson.org/v6/doc/upgrade/ , but unfortunately without success.

Thank you for your efforts

c-klinger, 2020-01-13:
Hi, i will have a look at this in the next days :)

c-klinger, 2020-01-14:
Hi @KiWyFox , can you check with branch https://github.com/c-klinger/esp8266-arduino-wifirgb/tree/feature/arduinojson6 ?
I was able to compile it with arduinojson 6.13.0 and the rest api and web interface worked well. Unfortunately i have no led stripe available to test the full setup.

KiWyFox, 2020-01-14:
Guten Abend Chris

besten dank für die schnelle aktualisierung des Codes auf ArduinoJson v6.1.3. 
Für mich auch nützlich anhand der Diffs in der readme.md die Änderungen nachzuvollziehen.

Der Code lässt sich problemlos kompillieren. Auch die Steuerung des LED-Streifen, API und Web-UI klappt wie gewohnt.

Recht herzlichen dank für deine super Unterstützung.

Good evening chris
thanks for the quick update of the code on ArduinoJson v6.1.3.
The code can be compiled easily. The control of the LED strip, API and web UI works as usual.

Thank you very much for your great support.

closed 2022-06-05 by c-klinger